fritzing/fritzing-app #4147: Cant simulate circuits

squaze, 2024-05-27:
Hello,
im new with the Fritzing program, so i dont know if its my fault, but if i try to simulate any circuit with arduino nothing happens, no led lights up.
I also tried the example curcuits you can choose from in the program, but there is the same result. 
Someone has any ideas?

Im using this version:
Version 1.0.2 
(bCD-1901-0-dbdbe34c 2023-12-20) 64 [Qt 6.5.3]

Operating System:
<Windows 11 pro >

failiz, 2024-05-27:
You cannot simulate Arduinos. For now, just operating point analysis are supported. In a few months we will add suport for transitory simulations. See the current capabilities of the simualtor: https://blog.fritzing.org/2022/06/27/Simulating-Circuits-with-Fritzing

If you want to simulate Arduino, you can add a thumb up in this issue: https://github.com/fritzing/fritzing-app/issues/4038

The example circuits should run without problems, but only the ones in the simulator folder (Open Example->Simulator), the other ones cannot be simulated. If they work, please, close this issue to avoid duplicates. If the simulator examples do not work, then it is an issue and you should attach a screenshot of the simulation result.

squaze, 2024-05-30:
Thank you for the fast answer, then everything is workig fine.

squaze, 2024-05-30:
Would be great to simulate Arduino in the Future

closed 2024-05-30 by squaze