fritzing/fritzing-app #4150: add hatched ground as ground fill option

ClemensGruber, 2024-06-23:
Some applications, e.g. touch buttons, require hatched ground instead of plain ground, but this is currently in Fritzing only possible as a [complex workaround](https://forum.fritzing.org/t/how-can-i-add-hatch-fill-for-a-project/258/4) editing the SVG and replacing the ground layer afterwords.

So it would be nice to have the option "hatched"  as ground fill, (perhaps with different hatching percentages) for ground fill. 

See also: 
- https://forum.fritzing.org/t/feature-request-hatched-ground-planes/19958
- https://forum.fritzing.org/t/how-can-i-add-hatch-fill-for-a-project/258/

KjellMorgenstern, 2024-06-26:
Instead of a solid ground fill, an element like this should be made available, right?
```
<svg width="200" height="200" xmlns="http://www.w3.org/2000/svg">
  <defs>
    <pattern id="hatch" width="10" height="10" patternUnits="userSpaceOnUse">
      <path d="M0,0 L10,10" stroke="#000" stroke-width="1"/>
      <path d="M10,0 L0,10" stroke="#000" stroke-width="1"/>
    </pattern>
  </defs>

  <rect x="10" y="10" width="180" height="180" fill="url(#hatch)" stroke="black" stroke-width="2"/>
</svg>

```
[image]

ClemensGruber, 2024-06-26:
Nice finding! Did not know that SVG has such "easy" opportunities to realize a hatched area! 

From Espressif's [Touch Sensor Application Note](https://github.com/espressif/esp-iot-solution/blob/release/v1.0/documents/touch_pad_solution/touch_sensor_design_en.md) > section [3.3.7. Ground Plane](https://github.com/espressif/esp-iot-solution/blob/release/v1.0/documents/touch_pad_solution/touch_sensor_design_en.md#337-ground-plane) a good default (at least for touch sensors) could be: 

> Typical hatching for the ground fill is 19% (5 mil line, 50 mil spacing).

For more flexible parameters (don't know if we need / wont them in Fritzing) I found [a screenshot from _DipTrace_](https://kicad-info.s3.dualstack.us-west-2.amazonaws.com/original/3X/0/b/0bb12f0f04668527f09a225b4867cad25b504c3f.png) in this thread https://forum.kicad.info/t/copper-pour-with-hatched-fill-instead-of-solid-fill/860/8 e.g. with `line width` and `line spacing` as editable parameter.

KjellMorgenstern, 2024-06-26:
Beware, just an example. Ideally it would work, and Qt (the framework we use) has improved their support for fill patterns quite recently (earlier this year) : https://www.qt.io/blog/qt-svg-not-so-1.2-tiny-any-more

And we would still have to support the various transformations the pattern undergoes, especially exporting to gerber.

TRST-P, 2024-08-24:
I would like to express my interest in this feature😁
For me specifically, I use an additive PCB printer to create PCB's and full ground fills are extremely time consuming and ink consuming.

KjellMorgenstern, 2024-08-24:
I guess you do some further processing to "slice" areas in the gerber file into lines.

Nevertheless, is there a nozzle diameter that would allow direct conversion?

TRST-P, 2024-08-24:
The software for the printer creates toolpaths from imported gerbers (when they are correctly formatted😜).
The nozzles for the printer come in sizes of 100, 150 and 250 micrometer.
Check out voltera.io for more info.