fritzing/fritzing-app #4153: Can't Find Parts

Objuankanob1, 2024-07-05:
I installed Fritzing on my system but when I go to start a project and search for a part, I get xxxx/OneDrive/Documents/Fritzing/bin/search-desktop-xxxxxx.fzb not found error.

HELP!

failiz, 2024-07-05:
There are several errors caused for saving the files or installing Fritzing in OneDrive or Dropbox. I would recommend to uninstall it and reinstall it in your local harddrive and saving the files there too and do not use OneDrive.

KjellMorgenstern, 2024-07-06:
@Objuankanob1 Could you please check the Version of Fritzing you are using? Is that an official build, downloaded from fritzing.org?

There are some recent reports of OneDrive being activated without many questions or explanation to the user:
https://www.neowin.net/news/windows-11-is-now-automatically-enabling-onedrive-folder-backup-without-asking-
permission/

Objuankanob1, 2024-07-08:
Okay, but every time I reinstall Fritzing is says its installing it on C:\Program files\Fritzing and when T run it, it says it can’t find the parts library. I cant seem to fine the syntax that installs it on the location it says it does: c:\programfile\Fritizing.

KjellMorgenstern, 2024-07-08:
Without a proper version information, we will have to close this issue.

closed 2024-07-08 by KjellMorgenstern