fritzing/fritzing-app #4154: Image scale is wrong when exported to Gerber

Phantomx2, 2024-07-08:
## Current Behaviour
When a PCB is exported to **Gerber** file the silkscreen images size are not matching the ones in fritzing. The images in the gerber file are always the same size no matter the size parameter in fritzing.

[image]
[Bug.zip](https://github.com/user-attachments/files/16132276/Bug.zip)

**Build:**
1.0.3

**Operating System:**
Windows 11

**Steps to reproduce:**
- Using the silkscreen image object, place an image on the PCB. 
- Export to Gerber format 
- Use **Gerbv** or another tool to view the Gerber file.
- Notice that the image size is wrong. 

## Expected Behaviour
The image in the gerber file should be the same size as in fritzing.

See forum post for more informations:
https://forum.fritzing.org/t/silkscreen-image-wrong-size/24214/4

KjellMorgenstern, 2024-07-08:
If applicable, an SVG can be used instead to avoid the scaling problem:

[image]

Phantomx2, 2024-07-08:
I am using an svg in the included file, it doesn't fix the issue for me.

KjellMorgenstern, 2024-07-08:
In that case the SVG you are using probably contains a base64 encoded PNG image. It is not a vector graphic, although it is stored in a SVG container.

Phantomx2, 2024-07-08:
Ho, I exported it from photoshop so you are probably right about that.