fritzing/fritzing-app #4157: Test images for PCB load image import

KjellMorgenstern, 2024-07-09:
## Problem
The image formats requirement that work best for importing into a Fritzing PCB are currently not documented, and quite restrictive.

## Proposed Solution
Fix issues #4156,  #4155 , #4154 and more. 

To verify this, it would be useful to have a set of test images.

Such images should be usable in the PCB view: They should be monochromatic, or at least easy to threshold, so that they can be etched in copper or used for a silkscreen.

This issue is intended to collect some of those images.

poetaster, 2024-07-10:
I only have svgs for the pcb view, which generally work. I'll add those here, but I think you mean images to place on the pcb?
[image]
[image]
[image]

That's just a couple I use.

KjellMorgenstern, 2024-07-10:
@poetaster Thanks a lot, yes, SVG work best, and the issue is mostly about raster images. But it won't hurt to include some SVGs for tests as well.