fritzing/fritzing-app #4158: 1.0.3 (appimage) is completely unusable on Linux Mint 23.1

NormanDunbar, 2024-07-19:
## Current Behaviour

I've paid for and downloaded th latest 1.0.3 release. Unfortunately, I received an "appimage" file, which i do not like, however, it appears to be completely unusable on my system. Any information I can give you to resolve the issue, just ask.

Also, I am unable to provide any Fritzing files as I cannot create any without errors. 

**Build:**
fritzing-1.0.3-l1995-a7811ba0-qt6.AppImage

**Operating System:**
Linux Mint 21.3 64 bit. (libfuse2 already installed.)

**Steps to reproduce:**

## Numerous problems, I'm afraid.

### Cannot load some parts at startup.

On execution, I get a popup on screen with the following text (copied directly from the popup):

```
Unable to find the following 4 parts:

'generic_shrouded_pin_header_10_100mil' at 'generic_shrouded_pin_header_10_100mil,fzp'
'mystery_part_sip_3_100mil' at 'mystery_part_3_100mil.fzp'
'generic_female_pin_header_2_100mil' at 'generic_female_pin_header_2_100mil.fzp'
'generic_ic_dip_8_300mil' at 'generic_ic_dip_8_300mil.fzp'
```
I click OK to continue. I see a blank project.

## Cannot open existing projects.

On attempting to open any of my existing projects, I get another popup:

```
Unable to open '/home/norman/FritzingPCBs/NormDuino/NormDuino.fzz': zip.open(): -1000
```

All of my existing projects result in the same problem, I'm not able to open any of them.

## Errors placing some parts.

I attempted to create a new project with a battery, resistor and LED. On placing the battery, I get yet another popup:

```
Error reading file /tmp/.mount_fritziNCWia3/fritzing-parts/core/battery-AA.fzp: file 'pcb/jumper_2_100mil_pcb.svg' for title:'Battery' and moduleID:'1000AFDF10011leg' not found.
```

The battery does appear to be placed -- it can be seen on the breadboard and schematic tabs. It can also be connected to the other components.

## Cannot save new projects.

I am unable to save my new projects!  I attempted to save a new project to `/home/norman/fritzingPCBs/1.0.3_Testing/1.0.3_Testing.fzz` and another popup appeared:

```
Cannot write file temp:

/home/norman/.local/share/Fritzing/Fritzing/fzz/090291069f88a146dca4234201f6d574/temp.xml
/home/norman/.local/share/Fritzing/Fritzing/fzz/090291069f88a146dca4234201f6d574/1.0.3_Testing.fzz

No such file or directory.
```

This is not what I was trying to save! Also, the file I did request *was* created, but with a file size of only 22 bytes. Running a `hexdump -C` on the file resulted in:

```
00000000  50 4b 05 06 00 00 00 00  00 00 00 00 00 00 00 00  |PK..............|
00000010  00 00 00 00 00 00                                 |......|
00000016
```

## Expected Behaviour
Obviously, I expected to be able to open my projects and continue to develop/improve them, and to save any changes made. :disappointed: 

## Questions:

* Is there a way to obtain a proper (!) download of the executable which is not an AppImage?
* I would attempt to build the application myself, but it appears that the build instructions are out of date. Is there anyone who can update them please?
* If this cannot be resolved, how do I obtain a refund?

Cheers,
Norm.

KjellMorgenstern, 2024-07-19:
1. The generic parts are supposed to be created on the fly. Looks like Fritzing failed to write them.
2. When the fzz (a zipped file) is opened, it needs to be extracted. A backup directory is created, that contains the extracted file, a lockfile, and a temp bin for electronic parts specific to this project. This seems to fail.
3. When saving the file, first a temp file is written, so that the original does not get destroyed before we have the new file completely on disk. This fails again.

All these things point to a single issue: Fritzing can not write files on your drive. There might be several reasons for that.

KjellMorgenstern, 2024-07-19:
Can you check: 
- Write permissions to the mounted device
- Write permissions to the /home/norman/.local/share/Fritzing
- Space left on device

Especially, also check ownerships, e.g. maybe an earlier version of Fritzing was used as root user?

NormanDunbar, 2024-07-19:
Good Afternoon @KjellMorgenstern 

Many thanks for your very prompt response. Thank you.

I have checked and you were correct, the root user owned `/home/norman/.local/share/Fritzing`. I have no idea how this happened---I never use root for my "user" apps. Interesting. Perhaps a Docker image might have been running as root, I suspect this to be the case.

I have given permissions back to "norman:norman" and all appears well. 

* I no longer the the popup at execution time.
* I can now open my old projects, and save them with changes.
* Dropping a battery no longer pops up an error.
* New projects can be saved successfully.

Many thanks. I will close this issue.

Cheers,
Norm.

KjellMorgenstern, 2024-07-19:
Reopening this a issue to improve the error message. I think we could show a clear error if .local/share/Fritzing is not accessible.

KjellMorgenstern, 2024-07-19:
This might also point to a more serious bug. @NormanDunbar You mentioned that the file you wanted to save indeed was created, but only contained 22 bytes. 
Now, assume the file already exists, is a huge project and of course without backup. If saving the temp file (point 3 above) fails, Fritzing should not attempt to write the file. Would it overwrite the original file with 22 bytes garbage?

NormanDunbar, 2024-07-19:
> Would it overwrite the original file with 22 bytes garbage?

I believe it did when I tried a couple of times to save the file.

Cheers,
Norm./

closed 2024-10-10 by KjellMorgenstern