fritzing/fritzing-app #4161: Ghost connection when moving a part onto an existing connection

KjellMorgenstern, 2024-08-01:
## Current Behaviour

When moving a connector in top of another connector that already has a wire attached,
then the wire is attached a second time. A  ghost connection is created between J1 and J2.
Deleting the wire resolves the issue, but until this is done, there is unexpected behavior.

**Build:**
Probably all versions, tested on Fritzing 0.9.3 

**Operating System:**
All

**Steps to reproduce:**
1. We start with a wire connected to pin 2 of J1,  same as #4160 
2. Now, add a second connector (J2) and position it on top of J1.
3. Move the connectors apart (mouse or keyboard, doesn't matter)
4. See that there is now a ghost connection between J1 and J2.

Prepare:
[image]

Add connector on top:
[image]

Move apart and highlight net (hold mouse on any point of the net)
[image]

## Expected Behaviour
J2 should not pick up the wire from J1.

KjellMorgenstern, 2024-08-01:
The effect is similar to #4160. 
While 4160 is more an order of execution error, this one (#4161) points to more generic problems. 
When a part is positioned, it must not pick up  wires that are already connected.

A more general question, not affecting the bugfix (for now we just don't pick up the already connected wire):

Q:What should happen if the wire end is dangling?
Should it attach it when a part is placed on top of the loose wire end?

If we don't connect them, it will be difficult to spot these "cold" connections. But the same applies to parts that are placed on top of each other.

vanepp, 2024-08-02:
> Q:What should happen if the wire end is dangling?
Should it attach it when a part is placed on top of the loose wire end?

I think it should connect. I believe the rule is that a male connection when moved on to a wire or female connection will connect and I think (but am not sure!) that a wire should be similar, so assuming the part is type male the connection should be made. Indeed that seems to be how 1.0.3 reacts now:

starting  from

[image]

move the header down and it connects to the wire.

[image]

it also connects the other direction

[image]

move the wire to the header and it connects

[image]

move the header (male pin) on to the breadboard

[image]

and it connects (male header pin moved to female breadboard pin)

[image]

but move the breadboard on to the header (breadboard female to header male) and it doesn't connect

[image]

no connection (because female breadboard moved to male header and that doesn't connect)  

[image]

I don't know why it works like this, I have always assumed there is a code reason why this needs to be this way, but that may not be correct.

failiz, 2024-08-02:
> but move the breadboard on to the header (breadboard female to header male) and it doesn't connect
...
> no connection (because female breadboard moved to male header and that doesn't connect)
> 

@vanepp , that is a different issue: https://github.com/fritzing/fritzing-app/issues/3765

> Q:What should happen if the wire end is dangling? Should it attach it when a part is placed on top of the loose wire end?
> 
> If we don't connect them, it will be difficult to spot these "cold" connections. But the same applies to parts that are placed on top of each other.

I think they should connect to avoid the cold connections or, if not, give a warning and let the user select which part will be connected to the wire (but if the cold connector is made is going to be difficult to be spotted by another user, maybe a new color for the "cold" connection?). Parts that are placed on top of each other should connect if one is male and the other is female (as it is done for the shields). If there are two male connectors and a wire, I would say that both connectors should be connected electrically to the wire, and when you move one part, the wire stays with the part that has the "mechanical" connection and thus the two connectors of the part stop being connected electrically.

Maybe a better solution could be to only allow one connection per connector as it happens in the real world: it is not possible to connect a wire in a hole of the breadboard that already has a component (except for screw connectors that could have a few wires). It will force users to use more the breadboard. It will also require that we allow to make Y cables allowing wires to be connected to the bendpoints.

And continuing with replicating the real world, we could just avoid having overlaps of parts in the same layer. I think we now have the breadboard layer, the parts layer and the shields layer. That will solve this issue (probably will trigguer other though)

KjellMorgenstern, 2024-08-02:
z-order seems to be the key here:
- breadboard over header -> no connection, the breadboard doesn't have anything on the backside.
- header on a breadboard -> connection
- perfboard over male connector... well, this gets interesting. Especially is you consider the long standing and valid feature request, to be able to work on a perf board from both sides.

closed 2024-10-10 by KjellMorgenstern