fritzing/fritzing-app #4163: adding a notepad and 3d studio

S-Curley, 2024-08-12:
i would like to see fritzing expanded to include the option to add a simple notepad for planning projects and a 3D moddeling software package like blender to create the housing for my projects the way friting can include the arduino IDE.

i think they would both be good additions to the fritzing studio.
what do you think?

KjellMorgenstern, 2024-08-12:
Certainly, Post-its are one of the most important tools in project planning, and they can be used in Fritzing. Hopefully we can add some more direct control (e.g. font size, colors) in the future.

[image]

But I think it would be quite limiting to use it as a Kanban or similar board.

S-Curley, 2024-08-14:
they aren't really that good for writing out a design brief , update/upgrade list, bill of materials and services needed to complete a project and calculations work-through though, just to name some stuff of the top of my head. 
they seem best used for circuit annotation, for me at least. 
having a notepad as a plug-in like the way the arduino IDE can be plugged in to fritzing would be better for keeping all the different files of a project in one place/folder without having to do any separate file management.
thats what i was thinking...am i missing an obvious answer to this due to a lack of knowledge in my new hobby?
i am just starting out in engineering as a hobbyist on the arduino platform.