fritzing/fritzing-app #4164: fritzing missing maximize/ minimize button

S-Curley, 2024-08-12:
behaviour:
fritzing appImage missing maximize/ minimize button from the top right of the app window.

ScreenShot:
[image]

fritzing appImage
<please attach a Fritzing file if possible. To attach it, change the extension of the file from fzz to zip (e.g., bug.zip instead of bug.fzz)>
 ive added it as an attachment but i dont know where it is in gitHub (my first time , i dont know how to use this site yet, sorry) i think the file is too big to share i have a message at the bottom of this window saying "File size too big: 25 MB are allowed, 90 MB were attempted to upload." im not sure what to do.

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]
/home/shaun/Arduino/Programs/fritzing-1.0.3-l1995-a7811ba0-qt6.AppImage

**Operating System:** and system specs
System:
  Kernel: 5.15.0-118-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Dell product: Wyse 7040 v: N/A serial: <superuser required> Chassis:
    type: 3 serial: <superuser required>
  Mobo: Dell model: 0080PM v: A00 serial: <superuser required> UEFI: Dell v: 1.25.0
    date: 04/08/2024
CPU:
  Info: quad core model: Intel Core i5-6500T bits: 64 type: MCP arch: Skylake-S rev: 3 cache:
    L1: 256 KiB L2: 1024 KiB L3: 6 MiB
  Speed (MHz): avg: 800 min/max: 800/3100 cores: 1: 800 2: 800 3: 800 4: 800 bogomips: 19999
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel HD Graphics 530 vendor: Dell driver: i915 v: kernel ports: active: DP-1,DP-2
    empty: DP-3, HDMI-A-1, HDMI-A-2, HDMI-A-3 bus-ID: 00:02.0 chip-ID: 8086:1912
  Display: x11 server: X.Org v: 1.21.1.4 driver: gpu: i915 note:  X driver n/a display-ID: :0
    screens: 1
  Screen-1: 0 s-res: 2560x1024 s-dpi: 96
  Monitor-1: DP-1 pos: primary,right model: Dell P1914S res: 1280x1024 dpi: 86 diag: 482mm (19")
  Monitor-2: DP-2 pos: primary,left model: Dell 1907FP res: 1280x1024 dpi: 86 diag: 482mm (19")
  OpenGL: renderer: Mesa Intel HD Graphics 530 (SKL GT2) v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
Audio:
  Device-1: Intel 100 Series/C230 Series Family HD Audio vendor: Dell driver: snd_hda_intel
    v: kernel bus-ID: 00:1f.3 chip-ID: 8086:a170
  Sound Server-1: ALSA v: k5.15.0-118-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Ethernet I219-LM vendor: Dell driver: e1000e v: kernel port: N/A bus-ID: 00:1f.6
    chip-ID: 8086:15b7
  IF: enp0s31f6 state: up speed: 100 Mbps duplex: full mac: <filter>
Drives:
  Local Storage: total: 465.76 GiB used: 60.66 GiB (13.0%)
  ID-1: /dev/sda vendor: Western Digital model: WD5000BPVT-16HXZT3 size: 465.76 GiB
    speed: 3.0 Gb/s serial: <filter>
Partition:
  ID-1: / size: 456.89 GiB used: 60.66 GiB (13.3%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: 29.8 C pch: 55.5 C mobo: 27.8 C
  Fan Speeds (RPM): N/A
Repos:
  Packages: apt: 2659
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/kicad-kicad-8_0-releases-jammy.list
    1: deb [signed-by=/etc/apt/keyrings/kicad-kicad-8_0-releases-jammy.gpg] https: //ppa.launchpadcontent.net/kicad/kicad-8.0-releases/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
Info:
  Processes: 229 Uptime: 40m Memory: 15.48 GiB used: 4.52 GiB (29.2%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Unknown python3.10 client inxi: 3.3.13

**Steps to reproduce:**
- Step 1:  obtain a copy of fritzing and set the permission bit
- Step 2:  open said copy

## Expected Behaviour
i expected there to be a Minimize/maximize button in the top right of the app window so i could resize the app to my monitor for ease of use.

S-Curley, 2024-08-12:
i will change my linux distro if that helps just tell me which one to choose.
please remember im a cleaner not a programmer. thanks

vanepp, 2024-08-12:
For the history of this see

https://forum.fritzing.org/t/main-window-issue/24542

For me on the same Mint linux release (but different hardware and only one monitor) running from the Fritzing download Appimage, Fritzing works the same as Windows with all three resize buttons available and Fritzing appears to operate normally. Yet for @mMerlin on Fedora, he sees just the dismiss button which seems odd to me but may just be a QT quirk.

S-Curley, 2024-08-12:
ive just tried the fritzing appimage on my second pc and it works fine so i think it may be an issue with my first pc.
here is the system info for that pc for comparison.
hope it helps fix the button bug.

System:
  Kernel: 5.15.0-118-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop Mobo: ASRock model: B450 Steel Legend serial: <superuser required>
    UEFI: American Megatrends v: P4.20 date: 08/03/2021
CPU:
  Info: 6-core model: AMD Ryzen 5 3600XT bits: 64 type: MT MCP arch: Zen 2 rev: 0 cache:
    L1: 384 KiB L2: 3 MiB L3: 32 MiB
  Speed (MHz): avg: 2327 high: 4123 min/max: 2200/3800 boost: enabled cores: 1: 2199 2: 2069
    3: 2373 4: 2053 5: 4123 6: 1995 7: 2200 8: 2172 9: 2204 10: 1868 11: 2808 12: 1867
    bogomips: 91195
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA TU116 [GeForce GTX 1660] driver: nvidia v: 535.183.01 pcie: speed: 2.5 GT/s
    lanes: 16 ports: active: none off: DP-1 empty: DVI-D-1,HDMI-A-1 bus-ID: 09:00.0
    chip-ID: 10de:2184
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: DP-1 res: 1920x1080 dpi: 26 diag: 2148mm (84.6")
  OpenGL: renderer: NVIDIA GeForce GTX 1660/PCIe/SSE2 v: 4.6.0 NVIDIA 535.183.01
    direct render: Yes
Audio:
  Device-1: NVIDIA TU116 High Definition Audio driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s
    lanes: 16 bus-ID: 09:00.1 chip-ID: 10de:1aeb
  Device-2: AMD Starship/Matisse HD Audio vendor: ASRock driver: snd_hda_intel v: kernel pcie:
    speed: 16 GT/s lanes: 16 bus-ID: 0b:00.4 chip-ID: 1022:1487
  Sound Server-1: ALSA v: k5.15.0-118-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel pcie: speed: 5 GT/s lanes: 1
    bus-ID: 05:00.0 chip-ID: 8086:2723
  IF: wlp5s0 state: down mac: <filter>
  Device-2: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: ASRock driver: r8169
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: d000 bus-ID: 08:00.0 chip-ID: 10ec:8168
  IF: enp8s0 state: up speed: 100 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Intel AX200 Bluetooth type: USB driver: btusb v: 0.8 bus-ID: 1-5:2 chip-ID: 8087:0029
  Report: hciconfig ID: hci0 rfk-id: 0 state: down bt-service: enabled,running rfk-block:
    hardware: no software: yes address: <filter>
Drives:
  Local Storage: total: 1.61 TiB used: 48.98 GiB (3.0%)
  ID-1: /dev/sda model: SATA3 120GB SSD size: 111.79 GiB speed: 6.0 Gb/s serial: <filter>
  ID-2: /dev/sdb vendor: Seagate model: ST500DM002-1BD142 size: 465.76 GiB speed: 6.0 Gb/s
    serial: <filter>
  ID-3: /dev/sdc vendor: Seagate model: ST500DM002-1BD142 size: 465.76 GiB speed: 6.0 Gb/s
    serial: <filter>
  ID-4: /dev/sdd vendor: Seagate model: ST3500413AS size: 465.76 GiB speed: 6.0 Gb/s
    serial: <filter>
  ID-5: /dev/sde model: SATA3 120GB SSD size: 111.79 GiB speed: 6.0 Gb/s serial: <filter>
  ID-6: /dev/sdf type: USB model: Mass Storage Device size: 29.72 GiB serial: <filter>
Partition:
  ID-1: / size: 108.98 GiB used: 44.71 GiB (41.0%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: N/A mobo: N/A gpu: nvidia temp: 38 C
  Fan Speeds (RPM): N/A gpu: nvidia fan: 38%
Repos:
  Packages: 2515 apt: 2514 flatpak: 1
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
Info:
  Processes: 351 Uptime: 1h 7m Memory: 31.26 GiB used: 4.18 GiB (13.4%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Unknown python3.10 client inxi: 3.3.13

good luck...

S-Curley, 2024-08-12:
sorry i didnt mean to close the issue.
im new...

KjellMorgenstern, 2024-08-12:
Did you try the keyboard shortcuts?

On GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy,
there should be several combinations:

Full screen <kbd>Windows</kbd><kbd>↑</kbd> 

Restore size <kbd>Windows</kbd><kbd>↓</kbd>

also this should work:

Toggle between Full screen and original size <kbd>Alt</kbd><kbd>F10</kbd>

Some programs also accept <kbd>F11</kbd> to toggle the size , but I think Fritzing doesn't.

Also, check this with a different program, not only with Fritzing.

mMerlin, 2024-08-13:
With the possible interactions, that might need 'check this with a different program that is also using Qt'.

S-Curley, 2024-08-14:
no i didnt try the shortcuts, sorry i didnt see that post before i did a full clean install of mint 22
i did check other programs and they were fine although i have no idea if they were QT or not. i dont know programming.
i think my version of mint 21.3 was corrupted thats all. 
i was talking to peter on the fritzing forum and he has also experienced buggy buttons maybe he can help with further investigations. 
i got fed up a decided that it must be sign from the universe to move to mint 22, save me losing my shit and breaking something. 
sorry i bailed on the problem so soon, i jut wanted to start my new hobby in arduino-tronics, and thanks for showing an interest in fixing the bug.