fritzing/fritzing-app #4166: ARM64 Installer for Windows 11 on ARM devices needed

Simplycissmus, 2024-08-19:
## Current Behaviour
The current Windows 64-bit installer for Fritzing is not compatible with Windows 11 on ARM devices. When trying to install Fritzing on a Surface Pro with ARM processor, the following error message appears:

[image]

**Build:**
1.03

**Operating System:**
Windows 11 on ARM (Surface Pro with ARM processor)

**Steps to reproduce:**
- Step 1: Download the latest Windows 64-bit installer for Fritzing
- Step 2: Run the installer on a Windows 11 ARM device
- Step 3: Encounter error message about missing Visual C++ Redistributable

## Expected Behaviour
An ARM64 version of the Fritzing installer should be available for Windows 11 on ARM devices, allowing for native installation without relying on x86 emulation.

Alternatively, if an ARM64 version is not feasible, clear instructions on how to run Fritzing on Windows 11 ARM devices or information about future plans for ARM64 support would be helpful.

graemecns, 2024-09-24:
I have  Windows 11 Snapdragon X elite machine. I have in stalled C++ directly by itself ok.
After Fritzing installed failed but the fritzing install wants to install C++ again but it fails and if i ask it not to install C++ it still fails due to C++ and damages the installed C++.
Is there a install version of Fritzing that Just installs Fritzing without C++ which I already have installed
Or is fritzing not compatable with new ARM machines?

KjellMorgenstern, 2024-10-04:
Which version of Windows 11 are you using?

On Windows 11 24H2, Fritzing might run out of the box with Prism:
https://learn.microsoft.com/en-us/windows/arm/apps-on-arm-x86-emulation

KjellMorgenstern, 2024-10-10:
We have updated the installer to a more recent version for Fritzing 1.0.4: https://fritzing.org/releases/1-0-4

vanepp, 2024-10-14:
I wonder if the issue is that there isn't a Microsoft Visual C++ 2015 Redistibutable  (which I think is required by Fritzing and is installed if not present by the installer) available for Arm. I think the Arm port is fairly recent and they may not have done older versions for it. This is from Win 10 on Intel

[image]

Checking the Microsoft site for what Visual C++ redistbutables (if any!) are available for Arm and installing one may fix the issue. I don';t know if Fritzing would use a later version or if it needs the 2015 version and won't accept a later one or not.

Altervayne, 2025-01-05:
Can confirm that despite Prism, fritzing will not install on Windows for ARM. If it isn't installed already, it attempts installing the x64 C++ Redistributable 2015-2019 and then proceeds to throw the following error:
[image]
This happens even if you install the redistributable beforehand, the installation refuses to go forward.

toolmanca1, 2025-01-20:
> We have updated the installer to a more recent version for Fritzing 1.0.4: https://fritzing.org/releases/1-0-4 I currently don't have any Windows ARM device available to check if that solves it.

I just checked installing 1-0-4 on Win 11 ARM64 and it does not install, same error as above.  I did donate to your great work, hopefully soon you'll have a way for me to run it on ARM64.

KjellMorgenstern, 2025-01-20:
We got a native Windows 11 ARM device, and also VMs running (needed for reliable testing).

We updated to the Visual Studio 2022 redistributables (the 2019 version is probably not available for ARM). Also, we'll package the DLLs with Fritzing, as many other programs do, although not recommended by MS. But I think this will resolve one of the most common steps that fails, as quite a number of user still have trouble when the Fritzing Installer asks for permission to run the Visual C++ Redistributable installer.

closed 2025-04-28 by KjellMorgenstern