fritzing/fritzing-app #4169: Add support for moving Bendpoints with the arrow keys

PickyBiker, 2024-09-11:
## Problem
Currently we can use the arrow keys to move parts and via’s on grid points, but that feature doesn’t apply to Bendpoints on the PCB or schematic. 

## Proposed Solution
Please make it possible to move Bendpoints around the grid using the up, down, left, and right arrow keys on both the PCB and Schematic.

KjellMorgenstern, 2024-09-12:
This is would be very useful. 

Right now, it is not possible to select bendpoints and big dots, they can only be dragged or clicked with the mouse.

Left click highlights the subnet.
Right click opens a context menu (e.g. turn into Via)
Double click deletes the bendpoint.

They can not be aligned, or moved or chosen with the keyboard.

Ideas:
- Bendpoints should become selectable, visualized with a dotted line rectangle or circle around it.
- When a wire is selected, <kbd>tab</kbd> moves along the bendpoints and endpoints of the wire.
- When a bendpoint is left clicked we still highlight the subnet, but also select the point.
- Holding Ctrl while clicking adds it to the current selection.
- If a bendpoint is selected it can be 
  + moved with arrow keys
  + removed with the delete key
  + menu button opens the context menu: turn into via (pcb view).
- If a bendpoint is selected in a group it can be
  + moved with arrow keys
  + removed with the delete key
  + aligned via the part menu

Related: https://github.com/fritzing/fritzing-app/issues/4170

KjellMorgenstern, 2024-10-01:
Also related: https://github.com/fritzing/fritzing-app/issues/3609