fritzing/fritzing-app #4177: Part numbers for wires in Breadboards

veng1, 2024-09-29:
I’d like to have a property like a part number property for a breadboard so I can specify the length of the wire in millimeters.

I have been able to compile the code to show the part number field but don’t see how to edit the field. How is that enabled?

## Proposed Solution
Add the ability to edit the wires with a part number field or similar.
Or, give me some guidance where the code to edit the part number field is located and I'll see if I can change it.  I'm primarily a Python guy, not cpp, so I need some fairly specific guidance.

KjellMorgenstern, 2024-10-01:
One starting point could be the "View'-> Color Wires By Length' feature, which indicates the length.

[image]

Another one could be https://github.com/fritzing/fritzing-app/issues/4179 to show labels on wires.

The Inspector is managed in the "HtmlInfoView"
I don't know about any existing use cases for the part number field for wires, and using it to code the the length, diameter or material makes some sense.

If a length is specified, I think Fritzing must then also use this as a constraint on the wire. It would be confusing to have a wire half across the breadboard that is specified as 5mm length. Even if we don't implement that constraint right now, the possibility is still a very good reason the use a distinct property for the length, and not re-use the part number.

Properties are part of the Qt framework (both, python or C++), see qobject.h

failiz, 2024-10-01:
To edit the field in Fritzing, the part needs to be a Capacitor item. Check capacitor.cpp and partfactory.cpp which makes this possible.

veng1, 2024-10-01:
To further elaborate, my "Breadboard" will actually be parts mounted on a DIN rail.  The Fritzing Breadboard diagram does a stellar job of illustrating this using photos to create svg images of the parts.  Consequently, the lengths of the wires are longer than the color codes which are used in the kits.  And this being a 120VA battery-up back circuit, the AC power wire should be the conventional colors while V+ should be red and V- should be black in this use case.  So, I think that part numbers are a reasonable solution.

A wire on a breadboard or any other real world circuit is a physical part just like a resistor, so I'd like to suggest this as an improvement to a future release.  However, I'll attempt to change the code for my temporary use.

Thanks for the guidance.

veng1, 2024-10-01:
I changed the Wire to a Capacitor item in partfactory.cpp but the field is still not editable.  What have I overlooked?

failiz, 2024-10-02:
You must add the property in the properties.xml file