fritzing/fritzing-app #4183: pcb view fabricate icon no longer displays cost

vanepp, 2024-10-05:
## Current Behaviour

The  pcb view fabricate button no longer displays the cost of the board in a window.

See forum post for the issue. Looks to have stopped working (which may be intentional!) with version 1.0.0

(https://forum.fritzing.org/t/fabricate-costs/25160)

**Build:**
Version 1.0.3 
(bCD-1995-0-a7811ba0 2024-05-30) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

see Current Behaviour

## Expected Behaviour

Display cost as 10.2 did unless the deletion was intentional.

KjellMorgenstern, 2024-10-05:
This might be a problem with the configuration. 
If you press the button sub-menu, see if any Fab is selected.

[image]

The price estimation currently only works with the Aisler Fab.

vanepp, 2024-10-05:
That indeed resolves the issue, I'll close this ticket. Thanks!

closed 2024-10-05 by vanepp