fritzing/fritzing-app #4187: 1.0.4 release on Win 10 appears broken.

vanepp, 2024-10-14:
## Current Behaviour

I installed the released version of 1.0.4 over 1.0.4B1 and the release version doesn't create a Fritzing icon on the Desktop (which is easily fixable) and doesn't associate the .fzpz of .fzz files with Fritzing (which as far as I can see isn't user fixable!) 

This is the file associations with 1.0.4B1 loaded (where they are correct) 

[image]

and the desktop with the Fritzing icon

[image]

so uninstall 1.0.4 release

[image]

and reinstall 1.0.4 release (with installer logging enabled)

[image]

accept the defaults (as I usually do) 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

It loads and works correctly (except for the lack of the desktop icon and the file associations.
reloading 1.0.4B1 restores correct operation as far as I can see (as does falling back to 1.0.3)

[image]

Here are the installer log files from a 1.0.3 install both in a zip file in case that helps.

[install-logs.zip](https://github.com/user-attachments/files/17358825/install-logs.zip)

**Build:**
Version 1.0.4 
(CD-2065-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]
**Operating System:**
Windows 10

**Steps to reproduce:**

See  ## Current Behaviour

## Expected Behaviour

Adds the desktop icon and associates the Fritzing files.

failiz, 2024-10-14:
I also just noticed that Fritzing 1.0.4 cannot be found when I click the start windows button and type Fritzing. This used to work before, and it is probably caused by not adding a shortcut in the start menu.
[image]

KjellMorgenstern, 2024-10-14:
The only difference between build 2051 and 2065 is that they are bundled with a different version of the installer:
fritzing-1.0.4-l2051-a8c6ef7c-win64_qt6.msi
fritzing-1.0.4-l2065-a8c6ef7c-win64_qt6.msi

You mentioned 1.0.4b1 . That version is not increased for each build, but it likely is
build 2049 ( fritzing-1.0.4rc2-l2049-904f8559-win64_qt6.msi ) or 
build 2046 ( fritzing-1.0.4rc1-l2046-1717c0a0-win64_qt6.msi )

Can you try again if installing build 2051 works? It should show up as Fritzing 1.0.4, too (not beta or release candidate).

I would then replace the current public build 2065 with the 2051 build.

vanepp, 2024-10-14:
I had reinstalled 1.0.4 release (The rc I am using looks to be build 2049 as the about box calls it rc1CD-2049-904f8559 2024-10-02 rather than fritzing-1.0.4rc2-l2049-904f8559-win64_qt6.msi but I assume the commit number is the important part.) After unistalling 1.0.3 and reinstalling 1.0.4 last night, it acted the same. I just now started the install again (with Version 1.0.4 (CD-2065-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3] already loaded) and selected repair, but it didn't add the shortcut or the file associations.

KjellMorgenstern, 2024-10-15:
I have replaced build 2065 with build 2088.  This should fix creating the shortcuts and file associations on windows.

vanepp, 2024-10-16:
Looks like the new release fixes the problem. I just installed and tested it and everything now appears to work fine in terms of file associations:

[image]

and .fzpz and .fzz files load in Fritizng when double clicked on so I will close this ticket as resolved. Thank you!

closed 2024-10-16 by vanepp