fritzing/fritzing-app #4195: Linear gradiant renders in Fritzing but doesn't export as svg.

vanepp, 2024-11-09:
## Current Behaviour

[Stromversorgung Breadboard-fixed.fzpz.zip](https://github.com/user-attachments/files/17689507/Stromversorgung.Breadboard-fixed.fzpz.zip)

(remove trailin .zip to recover the .fzpz)

This part has layerIds now, and a linear gradiant which renders in Fritzing 

[image]

but doesn't export as an svg

[image]

**Build:**
Version 1.0.4 
(CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

See ## Current Behaviour

## Expected Behaviour

Preferably the linear gradient should be copied in the svg. If that isn't possible then the linear gradient shouldn't render in Fritzing so the user will know it won't export.

failiz, 2024-11-10:
Related to #1269

KjellMorgenstern, 2025-04-29:
Gradient is defined as url=""...
When I copy the original definition into the exported svg, the gradient renders fine.
```
<defs>
  <linearGradient id="SVGID_3_" x1="-1765.6" x2="-1739.9" y1="-876.47" y2="-876.47" gradientTransform="matrix(-4.3711e-8 -1 -1 4.3711e-8 -856.22 -1615.5)" gradientUnits="userSpaceOnUse">
    <stop stop-opacity=".8" offset="0"/>
    <stop stop-color="#050505" stop-opacity=".7835" offset=".0623"/>
    <stop stop-color="#121212" stop-opacity=".7651" offset=".132"/>
    <stop stop-color="#292929" stop-opacity=".7457" offset=".2054"/>
    <stop stop-color="#484848" stop-opacity=".7256" offset=".2813"/>
    <stop stop-color="#707070" stop-opacity=".7051" offset=".3591"/>
    <stop stop-color="#A2A2A2" stop-opacity=".11" offset=".4386"/>
    <stop stop-color="#fff" stop-opacity=".01" offset=".5119"/>
    <stop stop-color="#B7B7B6" stop-opacity=".7064" offset=".6435"/>
    <stop stop-color="#868584" stop-opacity=".7509" offset=".7086"/>
    <stop stop-color="#5E5C5B" stop-opacity=".7944" offset=".7723"/>
    <stop stop-color="#3E3D3B" stop-opacity=".8367" offset=".8342"/>
    <stop stop-color="#282624" stop-opacity=".8775" offset=".8938"/>
    <stop stop-color="#1B1917" stop-opacity=".916" offset=".9502"/>
    <stop stop-color="#161412" stop-opacity=".95" offset="1"/>
  </linearGradient>
</defs>
```