fritzing/fritzing-app #4196: Of grid moving a component

chamilonster, 2024-11-12:
## Problem
when you are designing the PCB and moving the components so much time you want to make little adjust off grid

## Proposed Solution
 will be nice if you press ctrl while moving a component, you can move freely and off the grid your placement

KjellMorgenstern, 2024-11-12:
That is exactly how it is already implemented. Could you please include your exact version information of Fritzing, the Operating System and tge Operating System Version? Maybe also your input devices if you have anything  non standard.

chamilonster, 2024-11-12:
> That is exactly how it is already implemented. Could you please include your exact version information of Fritzing, the Operating System and tge Operating System Version? Maybe also your input devices if you have anything non standard.

I'm using Windows 10 and Spanish keyboard, it works when I move traces corners, but not on parts.