fritzing/fritzing-app #4198: Panel Board

veng1, 2024-11-14:
## Problem
A panel board is wired or built very similar to a bread board.

## Proposed Solution
A panel board is an assemblage of components although the components are typically bigger than, say, a resistor such as power supplies, circuit breakers, terminal blocks, etc.  These are often mounted to DIN rails or sometimes screwed to a back panel. Because of the pictorial nature of fritzing, it is an excellent way of documenting a panel board as it allows one to see where the wires and components go.

An improvement of fritzing would be to expand the concept of a wire.  In a panel board, a wire has several characteristics such a length, gauge, color, and endpoints and, of course, part number.  Endpoints identify where the wire starts and ends, for example, PS1-1 to CB1-1 or Surge1- Line to PS1-Line indicating a wire from a power supply terminal 1 to a circuit breaker or from a surge suppressor AC line to power supply AC line.

In designing and wiring panel boards myself I've found fritizing to be useful but would like to see the above characteristics of wires added along with a wire list generator.

A wire list should identify the wire and endpoints, perhaps with optional characteristics although personally I'd be happy with just endpoints and a part number.  This should be a printable table where each wire could be checked off both while wiring and as a quality control check.

KjellMorgenstern, 2025-04-29:
This is an interesting use case. A panel board would not use the PCB view, right? Only breadboard and, just maybe, the schematic. Asking, because there are requests to make it easier to create such "single view" parts.

veng1, 2025-04-29:
I have personally in my career as a circuit designer done 100s of PCBs and cannot see where one would go from a panel board to PCB.  So, no, I don't see where one would need the PCB view.

The schematic of a panel board is useful for trouble shooting or QC but the breadboard view is more useful in the construction phase.  In construction, one wants to see pictorial component location along with wire size, color and routing.