fritzing/fritzing-app #4200: Part editor: Error: "there are no copper layers defined in imported SVG-File"

bugith, 2024-11-17:
See #3096 
Importing a svg we get an error popup that directs to a known workaround but no link is supplied
[image]
[image]

## Current Behaviour
Text
<please attach a screenshot if possible>
<please attach a Fritzing file if possible. To attach it, change the extension of the file from fzz to zip (e.g., bug.zip instead of bug.fzz)>

**Build:**
<e.g. Version 0.9.6 (b) 64 [Qt 5.15.2]

**Operating System:**
Ubuntu 22.04

KjellMorgenstern, 2024-11-17:
The error messages looks correct, this SVG is not suitable to be used in Fritzing. Please also see my comment on #3096

KjellMorgenstern, 2024-11-17:
@bugith Can you check if the error message is still incomplete if you switch to English?

bugith, 2024-11-17:
Okay. The popup from the English GUI finely directs to https://fritzing.org/learning/tutorials/creating-custom-parts/providing-part-graphics/
So there is a translation problem
Thank you