fritzing/fritzing-app #4201: Unable to open the Inspector panel

JonSearle120, 2024-11-18:
Hello, I am a new user, still trying to find my way around the application.  I have created a simple design on a stripboard with a corresponding schematic, when I started I could see and use the Inspector panel at the bottom of the parts panel but now it has disappeared. I cannot find anything in the help sections to advise how to get it back, I have shut down and reopened the app and closed and reloaded my design, both with no success.  I'm afraid I may have deleted it by accident, I moved it off the screen using the mouse thinking I could find it and move it back, but its gone.  How can I regain my access to the Inspector panel?  This situation arose because I wanted to resize the stripboard but without access to the Inspector panel this is not possible.

Thankyou in anticipation of your help.

failiz, 2024-11-19:
Try to use the view or window menu. You can enable the Inspector view there. You probably closed it accidentally.

KjellMorgenstern, 2024-11-19:
[image]

Besides the Window menu, a less known option is the context menu of the menu bar. Just right click anywhere on it:

[image]

closed 2024-11-19 by KjellMorgenstern

JonSearle120, 2024-11-19:
Thankyou for your prompt reply.  Is the Context menu the drop-down list that appears when I select the Window tile in the menu bar?  If it is, I have done that and I see a drop down menu in which I can select; Minimize, Parts, Inspector, Undo History, and Layers.  Below is a link to a picture of my screen, you can see that I have selected Parts and Inspector, and the untitled sketch showing the problem.  I can see the Parts panel but not the Inspector panel which used to appear below the parts panel.  I have scrolled to the bottom of the Parts panel to  illutrate this.  Sorry to have to ask for further help, being a very new user I am very grateful for any advice.

Thankyou.
[image]

JonSearle120, 2024-11-19:
Further to my last post,  I've got it back!  Whatever I did I've now undone and so far it all seems to be working again.  Onward and upward, thankyou for your help.

Kind regards