fritzing/fritzing-app #4202: when simulated parts turn gray

MigMo, 2024-11-22:
[image]
## Current Behaviour
I'm using Fritzing for the first time and purchased v1.0.4. I have made some electronic projects and added the codes but, when I simulate all parts turn gray. Even here the push button example do the same. I have watched older versions of Fritzing being enabled Yet this one v1.0.4 dont have that enable button and it's not working. 

What is the way to activate Fritzing v1.0.4 Installed on Windows 11?

failiz, 2024-11-22:
The simulation does not support Arduino boards. Check the current capabilities in : https://blog.fritzing.org/2022/06/27/Simulating-Circuits-with-Fritzing

You can use the simulation examples: examples ->simulator 

There is work to add transitory simulations, bit Arduino support is more complex and it could take a while on being supported (and it is not guaranteed that we implemented it)

closed 2024-11-24 by KjellMorgenstern

MigMo, 2024-11-25:
[image]
Failiz, even without an Arduino the simulation grays out. The Arduino was just an example I selected off Fritzing, it was not that I made the project. Here is a very simple electronic example I put together to focus on my question of Fritzing Grayed out and not the project.

failiz, 2024-11-25:
In that example, one of the cables is not connected properly (indicated by the red dots at the ends). Try to connect the cable well. 
Notice that they are not grey now, but it is an open circuit, the led is off