fritzing/fritzing-app #4208: SPACEBAR = rotate component //OR// User-assignable shortcuts.

daxliniere, 2024-12-21:
Rotating a part into the desired orientation is a frequently-used operation.

In CAD software like Revit, pressing the space bar while an element or group of elements is selected rotates it in 90 degree increments.
Because of this history, I keep tapping the spacebar out of instinct. I understand there are already shortcuts to rotate, but from experience, they can't beat the easy-access position of the spacebar for far better ergonomics.

Alternatively, I wouldn't mind if we had a tab in Settings to create user-defined key mappings. (though obviously this would be a much larger undertaking.)

Thanks for your consideration,
Dax Liniere.

KjellMorgenstern, 2024-12-22:
The method spaceBarIsPressed() is referenced 14 times within the Fritzing code.  It is already used for moving the canvas, in combination with the mouse button. This is useful for devices that don't have a middle mouse button or scroll wheel.
Side note: For touchpads, this seems broken, or never worked, for touchpads. On multitouch-touchpads, however, the two finger scroll should work.

Many programs seem to use a toggle to switch into a rotate mode, often with a key combination that involves the <kbd>R</kbd>.  I think a rotate mode make most sense for 3d rotations.

Others use a combination of modifiers together with the arrow keys.
Gnome tools, and Inkscape, and Lightroom use the brackets <kbd>[</kbd> and  <kbd>]</kbd>, like Fritzing. 

I find the use of the spacebar a bit surprising. Which direction should it rotate?

Overall, I don't have the impression space bar is widely used for rotation, or that this would be a very intuitive or logical use. 

A key configuration is currently not planned. 

But there are tools, like Karabiner, that can map keys application specific. I find it useful to remap the Home key on macOS to got to start of the line, instead of start of the document.

daxliniere, 2024-12-22:
Understood, @KjellMorgenstern , thanks for taking the time to reply.

All the best,
Dax.

daxliniere, 2024-12-26:
I know you have given your response, but wanted to mention that EasyEDA has the same keybinding (space=rotate).

KjellMorgenstern, 2025-04-29:
If we ever add configurable keybindings, this should be considered. Closing for now.

daxliniere, 2025-04-29:
May I suggest you close it as Not Planned rather than Completed?

closed 2025-04-30 by KjellMorgenstern