fritzing/fritzing-app #4210: Gerber output doesn't match pcb view

vanepp, 2024-12-31:
## Current Behaviour

Load this svg 

[image]

as a pcb image in Fritzing. In pcb view it looks like this

[image]

but in the geber output it looks like this (displayed here by gerbv

[image]

the extra images are in the svg (displayed here in Inkscape) 

[image]

but they don't appear in the Fritzing pcb view but do appear in the gerber output. 

**Build:**

Version 1.0.4 
(CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

See ## Current Behaviour

## Expected Behaviour

The gerber output should match what appears in Fritzing pcb view (i.e. only the rectangle not all the extra rectangles which don't render in Fritzing pcb view.)

KjellMorgenstern, 2025-04-29:
The QSvg renderer probably removes the invisible elements at an early point. It would be tricky to show them in PCB view, also not in line with other renderers. So, we should remove these invisible elements when transforming the SVG into Gerber.

vanepp, 2025-04-29:
That should have the desired result (no extra elements in the gerber output) and thus be a good solution.