fritzing/fritzing-app #4214: Suppression of views in New Parts Editor

RAPTOR7762, 2025-01-12:
**Problem**
To me it is a bit troublesome to edit the .fzp file to suppress a view and zip the .fzp file and .svg files back into the .fzpz file

**Proposed Solution**
Can put checkboxes tin individual views that says “suppress view”

KjellMorgenstern, 2025-01-12:
What do you mean by "suppress view" ?

RAPTOR7762, 2025-01-12:
So taking a breadboard for example, the part will not appear in schematic and pcb views

KjellMorgenstern, 2025-01-12:
Breadboards have a specific implementation in Fritzing. For regular parts, this is not intended.

RAPTOR7762, 2025-01-12:
Yes but there are some parts that need pcb view to be suppressed

KjellMorgenstern, 2025-01-12:
If it doesn't have an electric connection, you can simply use the "Breadboard Image" part.

RAPTOR7762, 2025-01-28:
What do you mean by "Breadboard Image" part.

KjellMorgenstern, 2025-01-28:
There is a core part called breadboard image.

RAPTOR7762, 2025-01-29:
OK! Also sorry for unclear info earlier. So like for this part here
https://forum.fritzing.org/t/433mhz-wireless-remote-control-switch-dc-12v-10a-rf/26354/7

I suppressed PCB view (as you can't possibly mount it on a pcb board) after much difficulty. So it may be easier to just have a checkbox to suppress a view

vanepp, 2025-02-01:
He is basically asking for a parts editor option to reuse breadboard view as the pcb view  renderer like this

`    <pcbView>
      <layers image="breadboard/fc-51_breadboard.svg">
        <layer layerId="breadboard"/>
      </layers>
    </pcbView>`

because you need a renderer in pcb (Fritzing will error if you just delete the pcbView field in the .fzp file.) This creates a part with a blank pcb view (as the breadboard view won't render in pcb) which creates the desired effect. Basically he is asking for this to be a select box in parts editor (editing the fzp file is the only current way I know to do this.)

KjellMorgenstern, 2025-04-29:
I assume you want to create the part in Fritzing to create an image like the below. This illustration seems dangerours.
[image]
At least use different illustrations for low and high voltage connectors. 
Have a gap between low and high voltage parts of the circuit. 

Anyhow, the use case is basically the same as for any other module, the user just wants a component for a breadboard style wiring diagram (without the breadboard). Still, other users might be disapointed if they don't find connectors when the want to create a schemtic or PCB using this?

vanepp, 2025-04-29:
While I'm not really sure, I think what he is asking for an option to do this in the fzp file from the parts editor with a tick box. Basically suppress the pcb view by replacing it with the breadboard view (which has the effect of creating a blank pcb view and is something I often do by editing the fzp file.) That is what I did for him in this case to suppress the pcb view and I think what he is requesting. Fritzing appears to need a renderer for pcb view so you can't just delete the pcb section, but it recognizes that it shouldn't render the breadboard view in pcb and thus leaves it blank without complaining as it does when pcb view is deleted. 

`    <breadboardView>
      <layers image="breadboard/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_breadboard.svg">
        <layer layerId="breadboard"/>
      </layers>
    </breadboardView>
    <schematicView>
      <layers image="schematic/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_schematic.svg">
        <layer layerId="schematic"/>
      </layers>
    </schematicView>
    <pcbView>
      <layers image="pcb/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_pcb.svg">
        <layer layerId="silkscreen"/>
        <layer layerId="copper1"/>
      </layers>
`

replaced by this

`    <breadboardView>
      <layers image="breadboard/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_breadboard.svg">
        <layer layerId="breadboard"/>
      </layers>
    </breadboardView>
    <schematicView>
      <layers image="schematic/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_schematic.svg">
        <layer layerId="schematic"/>
      </layers>
    </schematicView>
    <pcbView>
      <layers image="breadboard/prefix0000_6fbdb41430491ba080df8d6e0bc872e1_11_breadboard.svg">
        <layer layerId="breadboard"/>
      </layers>
`

which will produce a blank pcb view for the part.

RAPTOR7762, 2025-04-30:
Yes that’s what I need

RAPTOR7762, 2025-05-02:
@KjellMorgenstern So basically ticking this checkbox

[image]

it'll change the .fzp file from

```xml
 <views>
  <iconView>
   <layers image="icon/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_icon.svg">
    <layer layerId="icon"/>
   </layers>
  </iconView>
  <breadboardView>
   <layers image="breadboard/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_breadboard.svg">
    <layer layerId="breadboard"/>
   </layers>
  </breadboardView>
  <schematicView>
   <layers image="schematic/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_schematic.svg">
    <layer layerId="schematic"/>
   </layers>
  </schematicView>
  <pcbView>
   <layers image="pcb/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_pcb.svg">
    <layer layerId="silkscreen"/>
    <layer layerId="copper0"/>
    <layer layerId="copper1"/>
   </layers>
  </pcbView>
 </views>
```

to

```xml
 <views>
  <iconView>
   <layers image="icon/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_icon.svg">
    <layer layerId="icon"/>
   </layers>
  </iconView>
  <breadboardView>
   <layers image="breadboard/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_breadboard.svg">
    <layer layerId="breadboard"/>
   </layers>
  </breadboardView>
  <schematicView>
   <layers image="schematic/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_schematic.svg">
    <layer layerId="schematic"/>
   </layers>
  </schematicView>
  <pcbView>
   <layers image="breadboard/prefix0000_0d35483fbd5d2862c6e9d1474958a60c_1_breadboard.svg">
    <layer layerId="breadboard"/>
   </layers>
  </pcbView>
 </views>
```

and when this checkbox

[image]

is ticked, it changes the .fzp file from

```xml
  <connector id="connector0" name="pin1" type="male">
   <description>pin 1</description>
   <views>
    <breadboardView>
     <p layer="breadboard" svgId="connector0pin" terminalId="connector0terminal"/>
    </breadboardView>
    <schematicView>
     <p layer="schematic" svgId="connector0pin" terminalId="connector0terminal"/>
    </schematicView>
    <pcbView>
     <p layer="copper0" svgId="connector0pin"/>
     <p layer="copper1" svgId="connector0pin"/>
    </pcbView>
   </views>
  </connector>
```

to

```
  <connector id="connector0" name="pin1" type="male">
   <description>pin 1</description>
   <views>
    <breadboardView>
     <p layer="breadboard" svgId="connector0pin" terminalId="connector0terminal"/>
    </breadboardView>
    <schematicView>
     <p layer="schematic" svgId="connector0pin" terminalId="connector0terminal"/>
    </schematicView>
    <pcbView>
    </pcbView>
   </views>
  </connector>
```

RAPTOR7762, 2025-05-21:
@KjellMorgenstern Any updates?

RAPTOR7762, 2025-11-13:
OK @KjellMorgenstern so back to this. Would it be a good idea to have this