fritzing/fritzing-app #4218: Running arduino code in the app

mwelford2, 2025-01-29:
## Problem
There is no way to run a simulation of your project. Obviously for simple stuff like a blinking led this isn't a huge issue but if you have a more complex project with lots of wires going everywhere just drawing it out isn't enough. It's hard to tell if an issue you make in real life is because of some wire not being connected correctly or the logic itself is wrong if you have 100 wires to sort through and check. On Fritzing this would be way easier since you can and bend points and keep the wires from overlapping so you see it all laid out nicely and you can easily tell if you simply mis-connected a wire, but theres no real way to check if you did that without having Fritzing simulate the project with code.

## Proposed Solution
There should be a way to run arduino code in the app. This would run something like Wowki simulator where you load the parts, make all the connections, and then write the code that would be loaded on to the virtual arduino. All the parts would work as they would in real life and you'd be able to easily see where your problem is.

Why should this be on Fritzing if its already done?
Put simply, as far as I've seen, it's only been done on online tools. That means if those sites, like wowki, go down or the company behind them discontinues the tool, all your projects are gone. You have no control over your own projects and files. If this were on Fritzing, all the files are on your computer, and even if Fritzing stops being developed you still have it downloaded and can edit projects and create new ones as long as you like.

RAPTOR7762, 2025-01-29:
The developers are now working on simulating arduinos.

Please, read the blog post, which specifies the capabilities of the software:

https://blog.fritzing.org/2022/06/27/Simulating-Circuits-with-Fritzing
https://www.youtube.com/watch?v=xtKA_Ufik1k 18](https://www.youtube.com/watch?v=xtKA_Ufik1k
Simulating microcontrollers is not possible in SPICE simulators (like fritzing) as there are no models for them. Reason being most of them don't, or the manufacture has issued a copyright notice on them.

All elements that are not being simulated are greyed out.

failiz, 2025-01-30:
This is a duplicate of https://github.com/fritzing/fritzing-app/issues/4038
Please, close this issue and write in the old one and give a thumbs up there too to indicate your interest in this feature. 
Arduino support is complex. As @RAPTOR7762 said, there are no SPICE models for them. Wowiki uses a digital simulation approach and struggles with analog simulations. The current plan is to improve the transitory simulation. Adding support for Arduinos will take a few years (if it is done).

RAPTOR7762, 2025-01-31:
Yep, clising the issue

RAPTOR7762, 2025-02-01:
Yes exactly! Writing simulations for all arduinos would be basically building another Fritzing from scratch...

closed 2025-02-04 by mwelford2