fritzing/fritzing-app #4220: Allow option for breadboard to be shown with higher contrast

jaltekruse, 2025-01-31:
## Problem
I recently taught a beginner class on electronics with breadboards and Fritzing was a great tool for creating diagrams to show the different projects we built throughout the class. One issue that I had is that we were working with full size breadboards, so often to highlight the relevant part of circuit I wanted the students to work on I would crop a piece of the overall board. Unfortunately the default view of the breadboard seems to be optimized to look nicer on a screen, with hard to see labels for rows and columns. By the time it was on  projector screen the numbers were impossible to see.

## Proposed Solution
Adding the option to increase contrast on some core components like the breadboard would be nice both for this presentation use case as well as accessibility for those with low vision. You could also consider making the default be this accessible version, or something a little higher contrast than it is today for a little more accessible default experience.

jaltekruse, 2025-01-31:
Also I do plan on learning to create components and try to get some submitted into the standard list. Some community created components were very useful for my class, but I was a little short on time so I needed to do one hack where I just GIMPed a different screen with one less pin into the final images because I couldn't find a component matching the pinout of ones I had bought. So I might be able to contribute a modified breadboard component, or at least modify the SVG, as I assume the component might be baked special into the software given its complexity.

KjellMorgenstern, 2025-01-31:
You probably mean the numbers on the breadboard, like "30" , "35" ?
[image]

I am not sure if these numbers should be taken as absolute. There is no magic in these numbers, the more important concept is that each column of 5 builds a connection. 

It is already possible to use a custom breadboard, although these are on the more tricky side to create. No changes in Fritzing needed.

jaltekruse, 2025-01-31:
@KjellMorgenstern Thanks for the quick reply! I do understand that the numbers shouldn't be taken as an absolute, and I had included reminder text to that effect on my slides. In part because I was using different buttons than the standard 4 pin ones that come in all of the beginner kits, so ours spanned 3 rows instead of 4, but the standard button in fritzing looked most visually similar to the ones I had I just used them in my diagrams.

Part of why the labels being darker would be useful is so that students can understand the orientation of the board when part of it is cropped, like seeing the column labels on the left or right side of a cropped view, or just being able to see which direction the numbers are counting up or down the breadboard.

This is an example from my slides, on a screen where you cannot read these numbers it is hard to tell what portion of the board I chopped off, especially for a newbie.

[image]

vanepp, 2025-02-01:
As @KjellMorgenstern  said it is relatively easy (at least for me, probably less so for you) to darken the numbers on the breadboard. Making a correct breadboard part is not trivial as there are a number of non documented requirements to have it work. That would create a new custom part to replace the breadboard that you would have to load but for slide creation (as opposed to general use) that shouldn't be a problem. If you want one make a request in the Fritzing forum parts request catagory (I read github relatively rarely and may not see it here.)