fritzing/fritzing-app #4221: Autorouted pcb fails DRC

vanepp, 2025-02-01:
## Current Behaviour

This sketch

[Gewächshaus Bewässerung.fzz.zip](https://github.com/user-attachments/files/18629521/Gewachshaus.Bewasserung.fzz.zip)

(remove the trailing .zip to get the sketch) which is described in this forum post

https://forum.fritzing.org/t/battery-expansion-shield-18650-v3/26791/18

when all traces are deleted then autorouted with production type set to professional (not custom!) and with cycles set to 500 because the default is too short, routing doesn't complete when it ends, does not pass DRC

[image]

**Build:**

Version 1.0.4 
(CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**

Windows 10

**Steps to reproduce:**

see ## Current Behaviour

## Expected Behaviour

DRC should pass after autorouting as far as I know. The rules are supposed to guarantee that.

KjellMorgenstern, 2025-04-29:
This sketch has many routing database issues:

[image]

Also, the connection between LED3 and R8 seems like a "cold" connection. See the suspicious shaped dot in the center:

[image]

When I pull it, it looks like this, it is not really connected:
[image]

KjellMorgenstern, 2025-04-29:
To resolve the autorouting / DRC missalingment, we need a better, simplified example.

vanepp, 2025-04-29:
Unfortunately this is the only example I know of. With the various issues in the sketch it may not be reproducible with a normal sketch. It may well be it won't fail with a correct sketch, but I will have a poke at it and see if I can find something. I think there is something odd about Han's system it seems to create corrupted sketches fairly regularly for some reason.