fritzing/fritzing-app #4222: Include micro:bit with Fritzing 1.0.5

RAPTOR7762, 2025-02-02:
## Proposed Solution
Maybe can add a micro:bit part (download the part here)

[Microbit.fzpz](https://forum.fritzing.org/uploads/default/original/2X/8/86a613dbfd59adffd2466d972a68eb75da464167.fzpz) (35.1 KB)

As well as the breakout board

[Microbit-breakout2-complete.fzpz](https://forum.fritzing.org/uploads/short-url/8VRRjXzSL29dfAHHKYz5VJNTLLy.fzpz) (26.8 KB)

HaraldRau, 2025-02-02:
I think you have to fork the parts and then trigger a pull request.
You can put the parts in the forum first, or are they already there?

RAPTOR7762, 2025-02-02:
It is stated in the forum post here:
https://forum.fritzing.org/t/improved-micro-bit-part/7288/11

RAPTOR7762, 2025-02-03:
OK, submitted issue in Fritzing-parts

closed 2025-02-03 by RAPTOR7762