fritzing/fritzing-app #4225: simulator error on netlist

Mikeluw, 2025-02-08:
When trying to simulate my circuit and Fritzing, I get the following error:
Errors:
stdout Note: No compatibility mode selected!
stdout Circuit: simulator netlist
stderr Error on line 3 or its substitute:
stderr qm1 base_q_m1 17 pnp_generic off
stderr could not find a valid modelname

Netlist:
Simulator Netlist
RM1a 40 17 130
QM1  base_Q_M1 17 PNP_GENERIC off
RM1b base_Q_M1 40 200
R1 2 6 220

.MODEL PNP_GENERIC PNP ()

.option savecurrents
.option interp
.OP
*.TRAN 1ms 100ms
* .AC DEC 100 100 1MEG
.END
 The simulator gave an error. When loading the netlist, some spice fields may be wrong. Please check them. If the parts are from the simulation bin report, the bug is in GitHub. I have already tried checking the parts in the simulation bin, but I still experience this issue.

failiz, 2025-02-09:
Could you uploads the circuit file? Compress it in zip so you can upload it

Mikeluw, 2025-02-09:
Thanks for your response. I hope you can access the file.

[ezyzip.zip](https://github.com/user-attachments/files/18726026/ezyzip.zip)

failiz, 2025-02-09:
Thanks for reporting the error, but there is a problem with the IR sensor. The IR sensor M1 is not correct as the its spice filed references connector0 and connector0 is not defined in the part. Thus, that node is missing and the simulator gives an error. Please, try to use core parts if possible such us the FZ-51. There is also an analog IR sensor in the bin "SIM".

The second issue is more difficult to fix as the simulator is vary basic at the moment. Most of your parts do not have spice fields and cannot be simulated and we cannot simulate Arduino. Please, take a look at the blog, which describes the current functionality of the simulator: https://blog.fritzing.org/2022/06/27/Simulating-Circuits-with-Fritzing

failiz, 2025-02-09:
And please, close the issue as it is not a bug in the source code of the program.

Mikeluw, 2025-02-10:
Thanks, I will look into your feedback.

closed 2025-02-10 by Mikeluw

KjellMorgenstern, 2025-02-10:
There might be another unrelated issue showing in ezyzip sketch:

There are some open connections, indicating a missing part:
[image]

This might be a custom part, containing errors that prevented it from being bundled with the sketch. Just guessing, it is hard to say for sure. But even if the part has errors, Fritzing should not silently drop it, but complain about it.

@Mikeluw Can you check what part was connected there, and maybe post a copy?