fritzing/fritzing-app #4226: Warning when ground fill can't connect due to no path

vanepp, 2025-02-16:
## Problem

In forum post 

https://forum.fritzing.org/t/fritzing-file-looks-good-but-manufactured-boards-have-open-circuits/27037

a user had a ground fill issue where 3 seeds were in fact not connected to the ground plane. In this sketch

[8085romboard.fzz.zip](https://github.com/user-attachments/files/18813338/8085romboard.fzz.zip)

(remove the trailing .zip to recover the .fzz file!)

 It turned out to be operator error as he had used autoroute and autoroute had crowded the the traces so badly that there wasn't an available path to connect the ground fill (but Fritzing didn't complain about it either if Fritzing knew the ground fill failed!) 

[image]

specifically here, IC 10 pin 1 is set as a ground fill, but as we see due to the autorouted traces being too close there is not a path for the ground fill to get to the rest of the grounds. So delete the fill then clean up the trace routing to clear space around the ground fill seed.

[image]

then run ground fill again

[image]

and now ground fill connects correctly as it has a path to the rest of ground. Indicating this was operator error rather than a bug thus this is an enhancement request. 

## Proposed Solution

It seems to me that Fritzing should able to detect that the ground fill to one of the ground fill seeds didn't succeed (that may or may not be correct as I don't know how ground fill works internally!) ideally an error message that the ground fill wasn't able to connect to a ground fill seed would at least prevent this person from ordering boards that turned out to be broken because the ground fill wasn't compete.

KjellMorgenstern, 2025-03-13:
It looks like in the first picture, the ground fill connection is on the bottom side of the through hole connector.

Of course, this could be a single sided PCB, so that doesn't invalidate the example

[image]

Even if we leave space, the ground connection might still be not as expected:

[image]

vanepp, 2025-03-19:
> It looks like in the first picture, the ground fill connection is on the bottom side of the through hole connector.

That is true, but the bottom layer also isn't connected to ground due to the same issue (no clearance for the trace because of other paths which is operator error!) 

bottom of the board (gerber output displayed by gerbv) the ground fill is connected to the copper (this is a double sided board) but does not connect to ground because of the lack of clearance for a path to ground. For new users this is likely (as in this case) to get missed resulting in boards that don't work. 

bottom layer gerber output no connection to ground due to no path

[image]

top layer gerber output also no ground connection due to no path and no connection to the copper fill either which won't matter because of the through hole plating. 

[image]

The suggestion to route the grounds is a good one as that would cause sufficient clearance to guarantee a path to ground, but an error message if a ground connection can't be made (assuming it can be detected easily) would be better I think. In this case the user used autoroute and ground fill and assumed all would be well (which I expect is a common expectation with new users!) which it wasn't.

KjellMorgenstern, 2025-04-29:
When calculating the ground fill, Fritzing knows the number of polygons.
We are safe if there is only one polygon. 
We could warn about any number of polygons > 1, since this would mean separated ground fills.

More than one ground fill polygon, disconnected. So we would warn:
[image]

vanepp, 2025-06-17:
This looks good to me, the main thing is to warn that the ground fill isn't as expected so at worst the user asks why they are getting the message and this should do that I think.

closed 2025-10-21 by KjellMorgenstern