fritzing/fritzing-app #4228: Part of breadboard image disappears when Fritzing breadboard is scrolled.

vanepp, 2025-02-21:
## Current Behaviour

This sketch

[heater.fzz.zip](https://github.com/user-attachments/files/18897159/heater.fzz.zip)

(remove the trailing .zip to get the sketch) 

using this new part

[100w-ptc-air-heater.fzpz.zip](https://github.com/user-attachments/files/18897175/100w-ptc-air-heater.fzpz.zip)

again remove the trailing .zip to get the .fzpz file. 

when scrolled out in breadboard at some point loses the connector and wires (but not the connectors) and I don't know why. All elements are normal as far as I can see, but the result is not 
Scrolling in works fine but at some point scrolling out parts of the image no longer render

[image]

circled in red here should be the red tab and wires in the image above. 

[image]

**Build:***

Version 1.0.4 
(CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

See ## Current Behaviour

## Expected Behaviour

Scrolling should work unless I have done something odd that I am not seeing in the svg.

KjellMorgenstern, 2025-02-21:
The complete image is 4.7 inches wide, while the viewbox is set to 3.85inches.
I am not sure why is is sometimes displayed and sometimes not, but adjusting the viewbox will certainly fix it.

Unrelated, there is a little dust, id=path80 , that is probably not intended.

[image]

vanepp, 2025-02-21:
Yep, operator error. I added the connector and forgot to resize the view box and didn't see that. Fritzing used to truncate anything not in the viewbox which makes this obvious it may be worthwhile to make that happen again. I will close this as resolved and thanks for the assist!

original svg with too small viewbox (now corrected)

[image]

the corrected part now scrolls correctly

[image]

[image]

closed 2025-02-21 by vanepp