fritzing/fritzing-app #4229: Add customizable DIP IC sockets

RAPTOR7762, 2025-03-02:
## Problem and proposed solution
In this forum post
https://forum.fritzing.org/t/set-of-standard-wide-ic-sockets/27388

Someone requested a customizable DIP IC socket. I told him that it is possible to use a generic IC, but he told me that it would be better to have IC sockets

KjellMorgenstern, 2025-03-03:
The forum post refers to

https://www.advanced.com/pdf/AIC_DIP_Sockets_Molded_16A_rev0.pdf

I think an open frame socket could work well, as it is easier to distinguish from an IC without a socket.

Also, it enables hacks like this:

[image]

So, Fritzing IC Sockets would look like two header rows, width bridges left and right, not further bridges... although in reality
many sockets have additional stabilization:

[image]

If we provide a minimal open socket, such additional stabilization can be quite easily added if desired for a custom part.

RAPTOR7762, 2025-03-03:
Hi Kjell,

For the D31A* relay part, there is a problem on my side

https://github.com/fritzing/fritzing-parts/issues/406