fritzing/fritzing-app #4230: Update Issue

ismayel21, 2025-03-05:
when using "check for updates" in the Help tab a message of "no new versions" appears. 

Fritzing version:
1.0.3

OS
MacOS Sequoia 15.3.1

RAPTOR7762, 2025-03-06:
I don’t think Fritzing is updated with the GitHub repo anymore, so it won’t update. If I’m wrong, please correct me

KjellMorgenstern, 2025-03-06:
Parts updates are disabled, because even minor issues when pulling changes via git would leave users without git knowledge stranded with a broken installation. Minor issues would be: non-fast forward commits, github timeouts, etc... this was the number one reason for broken fritzing installations on windows until ~2021.

But the button also checks the website for app updates, and I intend to repair this. The basic functionality isn't tricky,  but it needs to be done carefully, because a fix needs to be compatible with all versions of Fritzing.

KjellMorgenstern, 2025-03-06:
Closing in favor of #4024

closed 2025-03-06 by KjellMorgenstern