fritzing/fritzing-app #4231: completion issue in schematic

vanepp, 2025-03-08:
## Current Behaviour

This may be part misconfiguration rather than a bug, but I don't think so. It may not be easily fixable though either. 

First the two parts involved

[Raspberry-Pi-Pico-ups-module.fzpz.zip](https://github.com/user-attachments/files/19146065/Raspberry-Pi-Pico-ups-module.fzpz.zip)

[Raspberry-Pi-Pico-tht.fzpz.zip](https://github.com/user-attachments/files/19146068/Raspberry-Pi-Pico-tht.fzpz.zip)

the forum post which documents the issue

https://forum.fritzing.org/t/looking-for-raspberry-pi-pico-ups/27526

The intent here is to be able to drag the pico part in to the UPS board in breadboard and have it connect in schematic (pcb is suppressed and will use the pico pcb.) That works with some limitations 

Drag the CPU in to the sketch 

[image]

then move it over the female headers and it connects (also note it reports routing complete as it should.) 

[image]

That reflects with rats nest lines to schematic

[image]

and is routable but you can't connect a wire to the agnd pin and the routing shows 7 unrouted connections (the other 7 gnd connections for the CPU) but won't let you route any of them (presumably because only one connection is permitted to a net in schematic and ground is a bus.) 

[image]

**Build:**

Version 1.0.4 
(CD-2088-0-a8c6ef7c 2024-10-07) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

see ## Current Behaviour

## Expected Behaviour

Preferably schematic would show routing complete and we could add a wire to for the agnd connection although that may not be possible.

failiz, 2025-03-09:
I have now checked this issue carefully, but it may be the same as  #3974. If so, remember to close this one to avoid duplicates.

vanepp, 2025-03-13:
I agree this is probably related to https://github.com/fritzing/fritzing-app/issues/3974 and will therefore close this issue as it is being dealt with.

closed 2025-03-13 by vanepp