fritzing/fritzing-app #4233: Fritzing error when start simulation

lucianogardim, 2025-03-15:
Lubuntu 24.04

Fritzing caught an exception from simulationButton in event 3: bad_function_call

[image]

KjellMorgenstern, 2025-03-15:
Did you try our official Fritzing release from https://fritzing.org ?

lucianogardim, 2025-03-27:
> Did you try our official Fritzing release from https://fritzing.org ?

It's not free.

I wouldn't want to pay for something that might not work.

HaraldRau, 2025-03-27:
Fritzing runs flawlessly for me. I've tested it on Ubuntu 20.04, Ubuntu 24.04, and Debian 12. I'm currently using it productively on Ubuntu 20.04 and Debian 12.

HaraldRau, 2025-03-27:
I conducted a simple test. Here's my circuit.

[image]

[Beispiel_Sketch](http://fietsmeter.de/Simulation_Transistor.fzz)

It simulates the circuit without errors.
Should this topic be moved to the forum?

KjellMorgenstern, 2025-03-27:
The simulator works as described on https://blog.fritzing.org/tag/simulator/

closed 2025-03-27 by KjellMorgenstern

failiz, 2025-04-26:
@lucianogardim 
That error means that Fritzing cannot find the ngspice library. It is a bug of the package you used, not a Fritzing bug. You can use the official package or report the error to the maintainers of the package you installed.