fritzing/fritzing-app #4240: Dragging is still not working in Hyprland Wayland

anoduck, 2025-03-29:
## Current Behavior
The ability to drag is completely nonexistent.
A screenshot is attached, but rather pointless, because it does not provide any pertinent information.
Fritzing file is attached as well, but as with screenshot, does not provide any pertinent information.
Enabling debugging also did not provide any information.

**Build:**
Version 1.0.1 
(rcdebian/1.0.1-1+b5 2024-10-06) 64 [Qt 5.15.13]

**Operating System:**
Kali Linux Rolling Release Current

**Steps to reproduce:**
1. Install Kali Linux
2. Install Hyprland from kali package repository.
3. Install Frizing from kali package repository.
4. Experience the null of drag.

## Expected Behavior
When installed through flathub, Fritzing worked as expected. Which is more than likely what will be a temporary means to mitigate the issue. The reason why this was not preferred from the beginning is more or less preferential. Packages installed from Flathub place the system under more load, and take up more space.

[image]
[New_Test.fzz.zip](https://github.com/user-attachments/files/19516787/New_Test.fzz.zip)

KjellMorgenstern, 2025-03-29:
Please use our official release from fritzing.org .

closed 2025-03-29 by KjellMorgenstern

anoduck, 2025-03-31:
So, that way, regardless of my financial troubles, I have to pay for an application when it is not even certain if the application would be usable. Thanks.