fritzing/fritzing-app #4242: App Creates Folder on Documents on macOS

orome, 2025-04-23:
The app creates a folder in the user's Documents folder on macOS, and recreates this folder if the folder has been deleted. Only user-created files and folders should go in the Documents folder.

> The files in the user’s Documents and Desktop directories should reflect only the documents that the user created and works with directly. Similarly, the media directories should contain only the user’s media files. Those directories must never be used to store data files that your app creates and manages automatically.

KjellMorgenstern, 2025-04-23:
This is kind of a duplicated issue, but since there is no clear resolution, and this is for macOS, lets have a closer look.

There are several types of files, most common:
- Fritzing parts
- Fritzing sketches

Maybe we should add a dialog, to explain users that Fritzing needs a place to store *their* files, and that Fritzing needs them (the parts files at least) in a specific structure?

Fritzing follows the https://doc.qt.io/qt-6/qstandardpaths.html#StandardLocation-enum .
On Linux the XDG_DOCUMENTS variable can be used to override the documents folder location with an arbitrary one, see #4188  .

Are parts files, that the user imports to Fritzing, user documents or not?
My stand is yes, since you typically curate these parts carefully, to align with your electronics projects... it is your personal library of parts. 

Deciding if these files are "user created" or "user downloaded", and what difference that makes, seems a bit like a bike-shedding issue honestly.   Especially, since the decision to use this folder was already made, more than a decades ago... changing it now will for sure cause a lot of trouble.

Instead of the QStandardPaths::DocumentsLocation , should we use the QStandardPaths::HomeLocation to create the Fritzing subdirectory? Or should we hide these files away in one of the other locations?

The XDG_DOCUMENTS variable has the advantage, that users who modify it most likely know what they are doing. But what will happen if we make this an easy accessible option? 

Related issues: #3270  , #3576, #4075, #4188, #3216, #2289, #1938, #4018, #3454, #3327

closed 2025-04-23 by KjellMorgenstern

orome, 2025-04-29:
> Deciding if these files are "user created" or "user downloaded", and what difference that makes, seems a bit like a bike-shedding issue honestly. 

User created files are files that the user creates: those that the user explicitly creates. It's pretty well understood.

It's easy to think that whether one app dumps things in Documents is just shedding (arguably it is) but the defense doesn't scale. Documents should be something the user made and owns. The critical test is that the user should be able to move or delete anything in Documents without any side effects.

KjellMorgenstern, 2025-04-29:
@orome Thank your for the thorough explanation. If think deleting the Documents/Fritzing folder, and see if things break, is a good test. I will try to work it into our testing procedures.

If you drag and drop a part to Fritzing, and then exit Fritzing, it will ask if you want to keep that part. Would rephrasing this to asking if you want to keep the part in the Documents folder be explicit enough?

One known issue might be the part search: The search stores results in Fritzings config (not in the Documents folder). If the matching part was in Documents/Fritzing, and that folder is deleted, opening the search will produce an error about the dead link.
Something similar happens with the "Mine" bin. If the user adds a part, a reference will be stored in the "Mine" bin. If the part that the user added is deleted, again, Fritzing will show an error message about the missing part.

orome, 2025-04-29:
I'm not sure how much work it is but the ideal UX would be to offer the user a configurable location, with a default of `~/Library/Application Support` (which, for example, is where every other app on my Mac maintains this kind of stuff).