fritzing/fritzing-app #4245: Bad part and poor outline.svg fail spectacularly

vanepp, 2025-05-04:
## Current Behaviour

First the various files:

The original (broken) sketch (remove the trailing .zip to get the .fzz file)

[reloj_meru.fzz.zip](https://github.com/user-attachments/files/20025715/reloj_meru.fzz.zip)

then the corrected sketch where the corrected version of the part was added via a delete minus which then works correctly

[reloj_meru-fixed.fzz.zip](https://github.com/user-attachments/files/20025717/reloj_meru-fixed.fzz.zip)

then the outline.svg file (recovered from the .fz file of the sketch) it isn't in the correct format which may have a bearing on why things break so badly. 

[image]

and the forum thread where there is more information

https://forum.fritzing.org/t/empty-etching-and-gerber-files/28468

It looks like some combination of the outline.svg file and the AMS1777 5V regulator module.fzpz cause the issue, although just the two of them in isolation in a test sketch don't seem to break the same way.  Running DRC on the reloj_meru.fzz sketch causes this error

[image]

Attempting to export the gerber files from the same sketch causes this sequence 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

and creates this in an empty gerber output directory

[image]

which results in this gerber output displayed by gerbv

[image]

hopefully this will tell you folks something. 

**Build:**
Version 1.0.5 
(CD-2185-0-b9add9ea 2025-04-02) 64 [Qt 6.5.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

see current behaviour

## Expected Behaviour

Maybe better error message? The fault looks to be a broken part and an invalid outline.svg import and there may not be a cure other than correcting the files.

KjellMorgenstern, 2025-05-13:
Do these pass the CI tests on fritzing-parts ? 

Catching the error there would be independent from Fritzings build in guards. But it would be much faster to implement it on fritizng-parts, learn about the error, and the continue here in #4245 about a possible guard in Fritzing.

vanepp, 2025-05-14:
>Do these pass the CI tests on fritzing-parts ?

The Fritzing part fails FritzingCheckPart.py, I haven't yet tried fxp_checker_runner.py, but I expect it will fail. The part is from the net, not in core parts (and is broken.) It looks to be fairly hard to create the problem, I tried just the outline.svg and the broken part as the minimal case, but that sketch works and exports to gerbers successfully, so something further in the sketch is needed to cause the error. As a result I expect  this isn't going to be a common error, it looks to need both a custom shape and a broken part and custom shapes are fairly rare so some time to find a way to detect the error likely won't be an issue. I opened the issue only to alert you to the problem in case something can be done to detect it (if it isn't easy it may not be worth doing I expect as it is likely rare!)