fritzing/fritzing-app #4246: Trackpad pan right/left doesn't work anymore (macOS)

judepereira, 2025-05-04:
## Current Behaviour
I upgraded from 0.9.6 to 1.0.5 only because the Gerber export option started crashing. While that is fixed, I can no long pan sideways using the trackpad on my MacBook Air.

**Build:**
Version 1.0.5 
(CD-2176-0-b9add9ea 2025-04-02) Cocoa [Qt 6.5.3]

**Operating System:**
macOS 15.4.1

**Steps to reproduce:**
- Create a new sketch
- Go to PCB view
- Try to pan right/left - it doesn't work, but panning top/bottom continues to work

## Expected Behaviour
Panning sideways should work just as panning top/down.

KjellMorgenstern, 2025-05-04:
Please check the scroll settings in Fritzing preferences, probably it is using the settings from the legacy version.

KjellMorgenstern, 2025-05-05:
Probably a duplicate of #4224 , closing.

closed 2025-05-05 by KjellMorgenstern