fritzing/fritzing-app #4257: fehlende Bauteile

MoBaKai, 2025-06-21:
Moin ...
habe ich mein Geld aus den Fenster geschmissen ...
es fehlen mehre Bauteile nicht vorhanden ...
Habe eigentlich nur gutes von Euch gehört, nur ich muss mich leider einenes besseren belehren lassen ..
Na dann muss ich mir wohl was besseres als euer bescheidener dreck Laden suchen ...
Viel Werbung um nicht, gebt lieber auf ...

Gruß
Kai Gallenkamp

HaraldRau, 2025-06-22:
Guten Morgen,
ich wünsche Dir einen schönen Sonntag. Ich hoffe, dass Deinen Post keiner weiter kommentiert. Du hast ja alles sehr anschaulich beschrieben.
Gruß Harald!

RAPTOR7762, 2025-06-23:
@MoBaKai What components are missing? Go to the [fritzing forum](forum.fritzing.org)!

>Well, I guess I'll have to find something better than your crappy shop...
A lot of advertising for nothing, better give up...

What do you mean??

---

via google translate

@MoBaKai Welche Komponenten fehlen? Geh ins [Fritzing-Forum](forum.fritzing.org)!

>Na, dann muss ich mir wohl was Besseres suchen als deinen beschissenen Laden...
Viel Werbung für nichts, lieber aufgeben...

Was meinst du??

closed 2025-06-23 by KjellMorgenstern

KjellMorgenstern, 2025-06-23:
Aside the derogatory wording, there are some more problematic claims in this post.

One complain was about our advertisements. This is simple, we don't do any.
Our advertisement budget is zero. The user might have seen reviews or recommendations, non of which are paid either.

The second complain was about missing parts.
The user didn't post which parts are missing, from which manufacturer.  Here is an alphabetic list of manufacturers https://octopart.com/de/manufacturers . We don't claim any completeness. 

German translation
Abgesehen von der abwertenden Wortwahl gibt es in diesem Beitrag einige weitere problematische Behauptungen.

Eine Beschwerde betraf unsere Werbung. Das ist einfach zu beantworten: Wir machen keine.
Unser Werbebudget beträgt null. Der Nutzer hat möglicherweise Bewertungen oder Empfehlungen gesehen, von denen ebenfalls keine bezahlt sind.

Die zweite Beschwerde betraf fehlende Bauteile.
Der Nutzer hat nicht angegeben, welche Teile von welchem Hersteller fehlen. Hier ist eine alphabetische Liste der Hersteller https://octopart.com/de/manufacturers . Wir erheben keinen Anspruch auf Vollständigkeit.