fritzing/fritzing-app #4265: Property Hierarchy

UniquePete, 2025-07-26:
## Problem
I am in the process of creating a series of related parts. They are all identified with the same Family but in order to differentiate between the various parts, I have used variations on several properties, namely Type, Model and Variant. Within this Family, there are two Types, one with a single Model, the other with two distinct Models. The Type with a single Model also has only a single Variant. The Type with two Models has three Variants for each Model.

When I select a particular Type in the Inspector, I see all Models and Variants within the Family, not just those that are relevant for the particular Type. The problem with the present situation is that I need to include essentially redundant information in each of the properties to enable them to be distinguished from the same property on a different part.

As a specific example, consider the following:

<table>
<tr><th>moduleId</th><th>family</th><th>type</th><th>model</th><th>variant</th></tr>
<tr><td>Module 1</td><td>pressure sensor</td><td>transducer</td><td>NPT</td><td>3-Wire</td></tr>
<tr><td>Module 2</td><td>pressure sensor</td><td>probe</td><td>HDL300</td><td>2-Wire</td></tr>
<tr><td>Module 3</td><td>pressure sensor</td><td>probe</td><td>HDL300</td><td>3-Wire</td></tr>
<tr><td>Module 4</td><td>pressure sensor</td><td>probe</td><td>HDL300</td><td>4-Wire</td></tr>
<tr><td>Module 5</td><td>pressure sensor</td><td>probe</td><td>QDY30A</td><td>2-Wire</td></tr>
<tr><td>Module 6</td><td>pressure sensor</td><td>probe</td><td>QDY30A</td><td>3-Wire</td></tr>
<tr><td>Module 7</td><td>pressure sensor</td><td>probe</td><td>QDY30A</td><td>4-Wire</td></tr>
</table>

If all of theses parts are loaded and I select, say, Module 1, the Inspector will show the relevant properties, but will also offer (through the drop down menus) all of the options for all of the properties—the variant options will include both 2-Wire and 4-Wire, in addition to the only valid variant for the NPT model, 3-Wire. Perhaps what is more confusing is that there will be two 3-Wire variants, one for the HDL300 model and one for the QDY30A model, with no way of identifying which is which unless I include some additional identifying information in the variant name.

## Proposed Solution
In this situation, ideally, it would be nice if only relevant property options were displayed. This would presumably require that individual property tables be constructed with reference to the value of a chosen property, rather than just the family name as currently appears to be the case.

In this case, if I were to select Module 1, the family property would show 'pressure sensor', the type property would show 'transducer' and include 'probe' as an option, but the model property would simply show 'NPT', with no other options, and the variant property would show '3-Wire', without any other options. If I were to select Module 2, the family property would again show 'pressure sensor', the type property would show 'probe' and include 'transducer' as an option, the model property would show 'HDL300' and include 'QDY30A' as an option, and the variant property would show '2-Wire', with both '3-Wire' and '4-Wire' as options.

I imagine that there would need to be some sort of property hierarchy involved as, following the above example, while it would be entirely logical to offer the choice of type options when a part is loaded, we would probably not want to go loading all of the type or model options that might include a '3-Wire' variant if that variant property of a part were to be explicitly selected.

I understand that this approach might violate some other logic at play within the Fritzing operating environment, so this is as much a feasibility question as it is a request for enhancement.

UniquePete, 2025-07-26:
I guess it depends on how the selection mechanism was implemented. If, for example, selecting the 3-Wire variant resulted in loading all relevant options for any member of the family that had a 3-Wire variant, you should see options for any types or models for which there is a 3-Wire variant.

More broadly then, loading Module 1 would result in the presentation of options related to any member of the family that had type transducer, model NPT _or_ variant 3-Wire. That last condition would result in the probe option appearing under type, because there is a probe with a 3-Wire variant, and similarly the inclusion of HDL300 and QDY30A options under model.

If we were to choose a module with a 2-Wire variant, or explicitly select the 2-Wire variant option (if/when presented), then transducer would no longer appear in the type list nor NPT in the model list.

But now I see that there could be a difficulty in getting the type and model options for all members of a family listed if they don't have at least one variant in common... and the most obvious solution to that problem is pretty much what we currently have. OK, so maybe the selection mechanism currently under discussion could be provided as an option, available through a check box, like there currently seems to be to lock Placement, for example.

KjellMorgenstern, 2025-07-28:
Do you think the improvements suggested in #4258 and #4259 is could substitute this a "property hierarchy" ?

I think switching properties should be "strict". When changing a property, no other properties should change at the same time. This means many transitions will not be possible anymore.
That way, the Inspector properties will be less confusing, and the part search more powerful.

UniquePete, 2025-07-29:
I guess I've always used the Inspector as a second level parts search (when there were property options available). Initially, I wanted to create something like the options available for a resistor. I wanted to use a single icon for, say the HDL300 specifically, or perhaps more generally a probe or pressure sensor, then use the Inspector to select various characteristics, narrowing down the search as properties were selected. This is probably precisely the taxonomic structure that you suggest.

This whole issue has really only become more important (to me) as I create/use more parts in my illustrations. When I only had a dozen or so that I worked with, a 'flat' structure was no problem. But now that I have more than 100 parts, that's just my own parts,  to work with, some sort of structure has become more important.

A related issue then is the fact that individual parts, even if they are almost identical, all have their own icons (at least I haven't yet discovered to how to group several parts under just one icon and then just use the Inspector to select a particular part based on its properties), unlike something like a resistor or LED (in Core Parts), where you simply select the part then nominate the rating or colour to display the correct rendering of the relevant breadboard image.

As well as the three different wiring options on the HDL300, as a further example, I have three more (electrically identical) parts with 50% scaled breadboard images (because the part is relatively large compared to most other components in associated electrical circuits). The result is six individual icons (some of which are essentially identical—I did try to reference the same icon file when creating the parts, but the Parts Library still included multiple copies of the icon) when, really, only one would be necessary to select the required part, with the specific details of breadboard image size and wiring selected through the Inspector. I can, of course, currently switch between all of these parts through the Inspector, it's the existence of six icons in the Parts Library that's the issue here.

I do note that, unlike the situation with resistors and capacitors, for some reason there are distinct icons for N and P transistors, even though the two are interchangeable through the Inspector.

So this is all really just about creating a deeper (more structured?) search hierarchy, one way or another. What we currently have is probably just a two level hierarchy. We start with the parts (icon) search, then refine that with what is essentially a 'flat' (potentially exhaustive) search of properties.

This suggestion really came about not so much because I'm having trouble finding parts but because, as I create more parts, I find myself having to go back over previous parts to 'refine' their properties. This last time around I figured I'd collate all the properties of all of the parts that I'd created in the interest of arriving at some sort of concise and consistent usage. As I did this, I discovered the issue that we've been discussing.

Don't get me wrong though, I love Fritzing—it's a great product! I'm just looking at ways to streamline the way that I use it, which is primarily for the purpose of creating illustrations of 'high level' circuit configurations.