fritzing/fritzing-app #4274: Pan using touchpad

lastant, 2025-09-21:
## Problem
Currently there seems to be no way to pan the canvas using laptop's touchpad.. The zoom in/out using two fingers scroll works great, but pan seems to only work by holding mouse middle button.

## Proposed Solution
Since the Ctrl+Dragging combination is used for adding to the selection, maybe a combination Shift+Dragging could pan the canvas?

lastant, 2025-09-21:
Now I see the "Mouse Wheel Behavior" settings, and the Guess option works great, thanks. With Pure the pinch zoom jumps around. In Scroll Priority pinch-zoom and up-down pan work (weird, I think pan didn't work before), but left-right pan doesn't.

KjellMorgenstern, 2025-09-25:
Lets set the "Guess" as default for Windows