fritzing/fritzing-app #4275: Add generic pin headers and solder eyes

Gerd-St, 2025-09-21:
## Problem
- I required a 9 pin header THT but there is only a 3 pin or a 6 pin header available. My workaround was using 3 3 pin headers placed one in line with the others to simulate. 
- I wanted to include a solder eye for connecting a cable to my PCB and found nothing within the library. My workaround was to abuse either a VIA or a resistor with one end located outside of the board.

## Proposed Solution
Please add a freely configurable pin header starting from 1 pin as well as a solder eye to the standard library or make existing parts editable.

KjellMorgenstern, 2025-09-25:
This already exists since probably the first release of Fritzing.

[image]

closed 2025-09-25 by KjellMorgenstern

Gerd-St, 2025-09-27:
I have searched again and within Core/Verbindung are some so called "generic" parts, e.g.  "Generic female header 2 pin", but none of them is customisable for its pin count. Still no idea on how to create a pin header, eithewr mal or female with 9 pins...

[image]

KjellMorgenstern, 2025-09-27:
You want to use the first one, then choose how long it should be.

Gerd-St, 2025-09-27:
The field with the pin count is not editable, how can I create a header with a specific number of pins?

[image]
[image]

KjellMorgenstern, 2025-09-27:
This seems to be a UX issue. I am quite sure out tutorials show it, but I agree it is not intuitive. I created issue #4277 for this.

Gerd-St, 2025-09-27:
I replaced my workaround with 3x3 pin headers with your gerneric header, but when checking the layout I am getting errors eventually because of too big solder eyes. 

[image]