fritzing/fritzing-app #4278: Cursor disappears

Val3374, 2025-09-28:
My cursor disappears when I hover it over a component, how to fix this ?

[image]
[image]

**Build:**
Flatpak version of Fritzing (installed via Bazaar)

**Operating System:**
Bluefin (Version: latest-42.20250928.1)

**Steps to reproduce:**
- Open Fritzing in breadboard view
- Hover the mouse cursor over any component
- Cursor disappears until moved away from component

## Expected Behaviour
The cursor should remain visible when hovering over a component

KjellMorgenstern, 2025-09-28:
Which platform?

Did you try the official version from fritzing.org?

Val3374, 2025-09-29:
I’m on Bluefin (Fedora Silverblue 42) using Flatpak (Wayland, GNOME)
I didn’t install from fritzing.org because it requires a payment

Val3374, 2025-09-29:
I’m on Bluefin (Fedora Silverblue 42) using Flatpak (Wayland, GNOME).
I didn’t install from fritzing.org because it requires a payment, only the Flatpak from Bazaar.

Val3374, 2025-09-29:
> Which platform?
> 
> Did you try the official version from fritzing.org?

I’m on Bluefin (Fedora Silverblue 42) using Flatpak (Wayland, GNOME).
!I didn’t install from fritzing.org because it requires a payment, only the Flatpak from Bazaar.

KjellMorgenstern, 2025-09-29:
PS The official build is available for x86 platforms, not yet for ARM. There are no known issues with mouse pointers on these (otherwise we would have fixed #4090 already)

closed 2025-09-29 by KjellMorgenstern

its-mrarsikk, 2025-12-28:
This same issue happens for me on x86_64 with Arch Linux. I'm on the latest Fritzing from Flatpak. Can we reopen this?

its-mrarsikk, 2025-12-28:
@KjellMorgenstern that's paid